maragelis/ParadoxMqtt32 #8: Home assistant config

pengemetal, 2021-02-04:
Hi, 

Does anyone use HA integration? How could I config. I can see in Mqtt Explorer the following, but I can not see any sensor state or sensor mqtt message, only the arm status (in Homekit topic) :

[image]

Thx

maragelis, 2021-02-06:
You haven’t enabled the hassio=1 in the h file

pengemetal, 2021-02-06:
> that’s the default message when the controller starts. As soon as you arm disarm you should start getting messages on the hassio message queue.

I tried, but nothing happened :(
I only get message from motion/door while armed?

maragelis, 2021-02-06:
What panel are you using ?

pengemetal, 2021-02-06:
ESP32 devkit1, I can try a devkit3

maragelis, 2021-02-06:
No what alarm panel are you using ?

maragelis, 2021-02-06:
Just updated the code with some major updates. just re-flash with the new code and try again.

pengemetal, 2021-02-06:
SP5000 with remote  GSM security service (but it is not connect to the serial port

maragelis, 2021-02-06:
Thats my bench panel used to create this project. So you should be ok.

pengemetal, 2021-02-06:
Thank you for the code! I managed to see the sensor state once! :) After reconnect I didn't. I think my wifi is too far from the paradox. I will try to fix the problem

closed 2021-02-08 by maragelis