maragelis/ParadoxMqtt32 #10: error compiling

woody4165, 2021-02-05:
Hi and thanks for sharing your project.

I'm not really familiar with VSCode and PlatformIO, but I succeded to istall n Linux (on Windows I wasn't able).

While compiling I get this error

```
.pio/libdeps/esp32doit-devkit-v1/LittleFS_esp32/src/esp_littlefs.c: In function 'esp_vfs_littlefs_register':
.pio/libdeps/esp32doit-devkit-v1/LittleFS_esp32/src/esp_littlefs.c:208:9: error: unknown field 'utime_p' specified in initializer
         .utime_p     = &vfs_littlefs_utime,
         ^
.pio/libdeps/esp32doit-devkit-v1/LittleFS_esp32/src/esp_littlefs.c:208:24: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
         .utime_p     = &vfs_littlefs_utime,
                        ^
.pio/libdeps/esp32doit-devkit-v1/LittleFS_esp32/src/esp_littlefs.c:208:24: note: (near initialization for 'vfs.<anonymous>.access_p')
.pio/libdeps/esp32doit-devkit-v1/LittleFS_esp32/src/esp_littlefs.c:179:27: warning: missing braces around initializer [-Wmissing-braces]
     const esp_vfs_t vfs = {
                           ^
.pio/libdeps/esp32doit-devkit-v1/LittleFS_esp32/src/esp_littlefs.c:179:27: note: (near initialization for 'vfs')
Compiling .pio/build/esp32doit-devkit-v1/lib4d7/SPIFFS/SPIFFS.cpp.o
*** [.pio/build/esp32doit-devkit-v1/lib211/LittleFS_esp32/esp_littlefs.c.o] Error 1
```

What should I check?

Thanks

maragelis, 2021-02-06:
The webpage is still in development. It has to be safe and secure so it will take some time.
yes correct RX2/TX2 is GPIO16 and GPIO17.
You can use IP150 with esp32 but in Series, the IP150 has an output jack normally used for diallers, you will have to create a cable to connect to that. 
You can read the wiki from my other project  https://github.com/maragelis/ParadoxRs232toMqtt/wiki/Hassio-Home-Assistant-Node-red but there will be some differences. (Have to get around to fixing it some time)

woody4165, 2021-02-06:
> You can use IP150 with esp32 but in Series, the IP150 has an output jack normally used for diallers, you will have to create a cable to connect to that.

You mean the one named PCS 250/G ? 
I hope to understand what is the connector type to find somewhere in an electronic shop.

> You can read the wiki from my other project https://github.com/maragelis/ParadoxRs232toMqtt/wiki/Hassio-Home-Assistant-Node-red but there will be some differences. (Have to get around to fixing it some time)

Thanks

woody4165, 2021-02-06:
I tried connecting the ESP32 directly to the Serial port instead of the IP150 and all is working fine.
Now I've connected to the other port, but it doesn't work properly.

I only see this in MQTT 
[image]
the Arm / 0 appears and disappears for X seconds.

and this is the web panel configuration of the I/O

[image]
should I change something here?

maragelis, 2021-02-06:
I myself have never tried it as I don't have an ip150 but there are people that said it works. Have you tried arming and disarming the alarm system, So the the esp can get initial values.

woody4165, 2021-02-06:
Yes I have.

The messages I receive seems to be dirty
[image]

I have looked also at the issues of your ESP8266 project, but I haven't found a topic of people connecting the ESP to the IP150 output.

maragelis, 2021-02-06:
Event 220 sub 220 partition 38 ???? Sorry don't know what that is haven't seen it before

maragelis, 2021-02-06:
What panel are so you have is it of the sp range

woody4165, 2021-02-06:
It's an SP6000 and it I connect the ESP directly to the serial everything works fine

maragelis, 2021-02-06:
What's your ip150 firmware version. I know there are problems with version 4.

woody4165, 2021-02-06:
It's a very old one, 1.39.02

woody4165, 2021-02-06:
[image]

[image]

maragelis, 2021-02-06:
Sorry can't help. Open an issue on this project or to the 8266 one maybe someone else can help. If I get hold of a ip150 I will look into it.

woody4165, 2021-02-06:
Tomorrow I will try to go without IP150 and I will try to understand how it works, especially with NodeRed.

I will open, then, an issue, as you said, looking for someone who is using IP150 and this awesome project.

Thanks

maragelis, 2021-02-06:
If you are using home assistant there is a wiki article that everyone uses that might be of some help

woody4165, 2021-02-06:
I use HA, but for all my "automations" I prefer to use NR
Thanks, I will look into both!

woody4165, 2021-02-07:
Hi @maragelis 

I'm trying this morning to send command, receiving seems OK.

```
{
  "event": 0,
  "sub_event": 6,
  "partition": 0,
  "sub_eventD": "",
  "eventD": "Zone OK",
  "data": "ZONA X"
}
```

This is what I'm doing

publish with topic `paradox32CTL/in` and the message in json format (I use MQTT Explorer) is

```
{
 "password":"000000",
 "Command":"disarm",
 "Subcommand":"0"
}
```
and the status I receive is

```
{
  "status": "Problem connecting to panel"
}
```

all my password are 6-digit long, not 4, I don't know if this can be an issue...

Thanks

maragelis, 2021-02-07:
I am afraid yes I only coded for 4 digit codes. I will have to update the code for 6 digits

woody4165, 2021-02-07:
> I am afraid yes I only coded for 4 digit codes. I will have to update the code for 6 digits

Thanks, strange that if I put a 4-digit code, I get same message instead of wrong password or something similar...

maragelis, 2021-02-07:
The serial api doesn't give feedback if it's a wrong password or if it's a connection problem

woody4165, 2021-02-07:
I've recovered the Winload software to program the Paradox alarm and changed the user codes from 6 to 4 digits (changing also a parameter that tells to accept only 4-digits) and now it works fine.

Is it complicated to change the code to accept also 6-digits ?
If you need any help, let me know if I can do anything!

Thanks

maragelis, 2021-02-08:
Added support for 6 digit codes. set USE6DigitCode to true

woody4165, 2021-02-08:
Thanks @maragelis !

closed 2021-02-08 by woody4165