maragelis/ParadoxMqtt32 #17: Trying to get your code working on EVO 192

pisabell2, 2021-03-04:
Hello,

I have an EVO 192 Paradox panel and I have recently started playing with the D1 mini. For the latter, I used EspEasy, so I am not really familiar yet with the Arduino IDE.

Yesterday I downloaded Arduino IDE 1.8.13 and I tried to compile your code with it. I immediately ran on the following error:

    "StaticJsonBuffer is a class from ArduinoJson 5. Please see arduinojson.org/upgrade to learn how to  upgrade your program to ArduinoJson version 6"

I am not comfortable trying to perform that upgrade myself. Should I try to locate an earlier version of the Arduino IDE?

Many thanks in advance for your help.

pisabell2, 2021-03-04:
Please forget my (stupid) question above regarding the ArduinoJson version. I just realized that I can select whathever version I need in the IDE.

I have a hardware question in need of an answer before I can proceed with testing the comminication between my panel and the D1 mini. My understanding is that the pinout on the paradox panel is  as shown in the following picture:

[image]

Now, is it sufficient for me to connect to Tx and Rx on paradox to tx and Rx on D1 mini (perhaps with inversion), or do I need to connect paradox GND to D1 GND and Paradox Aux+ to some D1 pin?

Many thanks.

maragelis, 2021-03-05:
Aux+ provides 12v dc do not connect it to your D1 without a stepdown to 5v you will fry it.

pisabell2, 2021-03-05:
Thanks for your response. I assume then that if I feed my D1 from a
separate power supply, I only need to connect RX and TX, right?

pisabell2, 2021-03-05:
A little bit of progress, here. I was able to flash my D1 mini and set up my wifi and mqt server, Now, when I connect the D1 to the RX/TX pins of my EVO 192, I do receive confirmation that the serial connection is getting successfully established: a new mqtt topic "paradoxdCTL" is created and the following statuses get posted: "Paradox connected", "Firmware: SYSTEM_PARADOX_2.2.3".
Unfortunately, I cannot get anything more to be published. 
I was expecting the 37-byte status message to be automatically posted at regular intervals: is this what is supposed to happen? I also tried to post a request of the form:
    paradoxdCTL/in/"password:mypassword,Command:panelstatus,Subcommand:0"
but I did not get any response.

Any suggestions? Many thanks in advance.

maragelis, 2021-03-06:
If I remember correctly the evo series does not work with 37byte messages hens this project won't work.  But you can try. 

Connect ground from wemos to panel. You need the common ground connection. 

Chech that you have cross connection rx->tx tx->rx. 

If you still don't get live events check if the panel is actually sending 37byte messages.

pisabell2, 2021-03-06:
Thanks for your feedback. 
If I assume that the Paradox pinout is the same as in the picture I posted 2 days ago (e.g. Rx above Tx), then I need to connect D1-Rx with Paradox-Rx and D1-Tx with Paradox-Tx in order to get any response whatsoever. I suppose, then, that the pinout on my EVO 192 has its Tx above Rx.

In any event, when my D1 is connected with its Rx on the top pin of the Paradox, its Tx on the next pin and the GND on the third pin, my MQTT broker does receive the following MQTT message:

{"status" : "Paradox connected"
 "firmware" : "SYSTEM PARADOX_2.2.3"}

Thus, the serial connection is at least working to some extent. Unfortunately, I am not getting anything more. Is your program supposed to automatically broadcast the 37-byte message at regular intervals?

I am not sure where I can find the relevant info about the EVO 192 messages, but I will search for it.

Many thanks!

maragelis, 2021-03-06:
No you have to connect rx to tx and tx to rx. It's a cross connection

pisabell2, 2021-03-06:
I understand what you are saying, but I don't inderstand how it is possible that:

1) when I cross the wires (assuming the Rx pin is indeed above the Tx pin of the Paradox connection) I am not getting any response whatsoever, but

2) I do get the following mqtt message when I do not cross them:

{"status" : "Paradox connected"
"firmware" : "SYSTEM PARADOX_2.2.3"}

Isn't it possible that the EVO 92 is wired differently?

maragelis, 2021-03-06:
Could be

pisabell2, 2021-03-06:
I am not yet familiar with using the Arduino IDE.

I used ESPEasy for  a couple of other applications (for which the code what already available from the ESPEasy community). I saw that within ESPEasy, one can monitor over the Web a log of program running on the D1 mini . 

Is there any way to get a log of what is going on between the D1 Mini and my alarm panel?

ezequiellop, 2021-05-17:
I have an esp12e. does this project help me?
Where can I download the code to flash the esp12e?