maragelis/ParadoxMqtt32 #18: WifiManager package not available

CheeseE, 2021-04-23:
The package tzapu/WiFiManager@^2.0.4-beta is not available anymore, compile issue appears.

maragelis, 2021-05-04:
It does exist its just not available under linux/macos

tasosstr, 2021-05-08:
A BIG THANKS TO Dear MARAGELIS ! ! ! !

I had the same problem and what i did it for fix it and upload over the Serial (usb)

[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1 ;nodemcu-32s ;
upload_port = COM3
framework = arduino
lib_deps = 
	knolleary/PubSubClient@^2.8
	ottowinter/ESPAsyncWebServer-esphome@^1.2.7
	luc-github/ESP32SSDP@^1.1.1
	bblanchon/ArduinoJson@^6.17.2
	https://github.com/tzapu/WiFiManager.git

tasosstr, 2021-05-09:
For the first time you will be connect the ESP32 (even after power recover) you should be send at least a command like
{
 "password":"xxxx",
 "Command":"panelstatus",
 "Subcommand":"0"
}

for send the status of inputs, other wise you will never see the status of each zone