maragelis/ParadoxMqtt32 #19: Arm in Hassio

pengemetal, 2021-05-04:
Hi,

I can see in Home assistant every motion sensors, but I can't arm the alarm sysem remotly.  I think the problem is /Arm topic returns 0 (or 1) Home assistant needs "armed", "disarmed" ..etc. Could anyone help with a HA config wich works? Thank you!

maragelis, 2021-05-04:
its not 0/1 the topic is paradox32CTL/hassio/Arm/0 where 0 is the partition number 
At that topic the values are  "armed", "disarmed"

pengemetal, 2021-05-04:
Thx for your answer! It is very interesting, I am listening paradox32CTL1/hassio/Arm/0 and it seems to be an empty message. I use Mqtt explorer and I can see the same. Is it possible Paradox doesn't send the status?
[image]

maragelis, 2021-05-05:
Click on ARM 
[image]

pengemetal, 2021-05-05:
It is empty 
[image]

maragelis, 2021-05-05:
are you able to arm and disarm the system ?
Try doing it manually from the keypad it should update the arm status.

pengemetal, 2021-05-05:
Oh, thank you very much! It needs a manually arming first! Works fine in HA too! :)
[image]

maragelis, 2021-05-05:
ok good job

closed 2021-05-05 by maragelis