maragelis/ParadoxMqtt32 #21: Instructions for opening in Arduino IDE?

sparkydave1981, 2021-06-05:
Hi,

I just downloaded the current version to try and migrate from the old ESP8266 version however there are no files in the package that the Arduino IDE will open.  I'm not exactly proficient with Arduino so can someone please help me with some instructions?  Is there something I need to do so the IDE will open these files?  I'm using the latest (1.8.15) IDE.

Thanks,

Dave

sparkydave1981, 2021-06-08:
I tried creating a file called 'ParadoxMqtt32-0.0.1.ino' and putting it in the project directory with the following code (after a Google search) but trying to compile it gives the following error:

file code:
```
void setup(){}
void loop(){}
```

error:

```
C:\Users\David\Documents\Arduino\ParadoxMqtt32-0.0.1\src\main.cpp:2:18: fatal error: main.h: No such file or directory
compilation terminated.
exit status 1
Error compiling for board WEMOS D1 MINI ESP32.
```
I'd really appreciate some guidance on this one please.

tasosstr, 2021-06-14:
This code are for use in Platform.io with VScode ( much better editor from arduino ...)

See here https://www.youtube.com/watch?v=dany7ae_0ks  how to use it.

Enjoy.

sparkydave1981, 2021-06-21:
> This code are for use in Platform.io with VScode ( much better editor from arduino ...)
> 
> See here https://www.youtube.com/watch?v=dany7ae_0ks how to use it.
> 
> Enjoy.

Thank you for letting me know, watching now.  Hopefully the Wiki for this Git gets updated to tell people the process...

tasosstr, 2021-06-22:
> 
> 
> > This code are for use in Platform.io with VScode ( much better editor from arduino ...)
> > See here https://www.youtube.com/watch?v=dany7ae_0ks how to use it.
> > Enjoy.
> 
> Thank you for letting me know, watching now. Hopefully the Wiki for this Git gets updated to tell people the process...
> 
> I now get a failure to compile due to missing wifimanager:
> 
> ```
> Library Manager: Installing tzapu/WiFiManager @ ^2.0.4-beta
> Warning! Could not find the package with 'tzapu/WiFiManager @ ^2.0.4-beta' requirements for
> your system 'windows_amd64'
> ```
> 
> Yet I'm not using an AMD machine to compile this code, it's a Microsoft Surface Pro (Intel chip).

I had the same problem and what i did it for fix it and upload over the Serial (usb)

Find the platformio.ini on the left and replace the code with the bellow if you need serial upload too or replace the wifimanager line with this https://github.com/tzapu/WiFiManager.git

full code

[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1 ;nodemcu-32s ;
upload_port = COM3
framework = arduino
lib_deps =
knolleary/PubSubClient@^2.8
ottowinter/ESPAsyncWebServer-esphome@^1.2.7
luc-github/ESP32SSDP@^1.1.1
bblanchon/ArduinoJson@^6.17.2
https://github.com/tzapu/WiFiManager.git

sparkydave1981, 2021-06-23:
Thanks for your help, I am a step closer but now getting a file not found error:

```
Library Manager: Installing git+https://github.com/tzapu/WiFiManager.git
FileNotFoundError: [WinError 2] The system cannot find the file specified:
  File "C:\Users\David\.platformio\penv\Lib\site-packages\platformio\builder\main.py", line 177:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\David\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 591:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\David\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\David\.platformio\platforms\espressif32\builder\main.py", line 225:
    target_elf = env.BuildProgram()
  File "C:\Users\David\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "C:\Users\David\.platformio\penv\Lib\site-packages\platformio\builder\tools\platformio.py", line 62:
    env.ProcessProjectDeps()
  File "C:\Users\David\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "C:\Users\David\.platformio\penv\Lib\site-packages\platformio\builder\tools\platformio.py", line 140:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\David\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "C:\Users\David\.platformio\penv\Lib\site-packages\platformio\builder\tools\piolib.py", line 1062:
    project.install_dependencies()
  File "C:\Users\David\.platformio\penv\Lib\site-packages\platformio\builder\tools\piolib.py", line 898:
    lm.install(spec)
  File "c:\users\david\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 48:
    pkg = self._install(
  File "c:\users\david\.platformio\penv\lib\site-packages\platformio\package\manager\library.py", line 81:
    return super(LibraryPackageManager, self)._install(
  File "c:\users\david\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 97:
    pkg = self.install_from_url(spec.url, spec, silent=silent)
  File "c:\users\david\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 141:
    vcs = VCSClientFactory.new(tmp_dir, url)
  File "c:\users\david\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 54:
    obj = getattr(sys.modules[__name__], "%sClient" % type_.title())(
  File "c:\users\david\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 137:
    self.configure()
  File "c:\users\david\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 146:
    result = proc.exec_command([cls.command, "--exec-path"])
  File "c:\users\david\.platformio\penv\lib\site-packages\platformio\proc.py", line 117:
    p = subprocess.Popen(*args, **kwargs)
  File "C:\Users\David\.platformio\python3\lib\subprocess.py", line 951:
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Users\David\.platformio\python3\lib\subprocess.py", line 1420:
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
================================================================================================== [FAILED] Took 1.29 seconds ==================================================================================================
The terminal process "C:\Users\David\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.
```

tasosstr, 2021-07-14:
Hello friend, 

sorry for late reply.

it looks like you haven't install the python ?

Another good place for you are this https://randomnerdtutorials.com/vs-code-workspaces-esp32-esp8266/

sparkydave1981, 2021-07-19:
I have installed the following Extensions:

C/C++
PlatformIO IDE
Pymakr
Python for VSCode

I also installed NodeJS on my PC as it is a prerequisite for Pymakr.

When I try to compile I still get the same error.

sparkydave1981, 2021-08-19:
Has someone got any advice for me here?  I am really looking forward to using this new version of Paradox interface but just can't get the VSC side of things working correctly to program the ESP

scvvuuren, 2021-08-28:
@sparkydave1981, I have the exact same challenge that you do. I either get the `.pio/libdeps/esp32doit-devkit-v1/WiFiManager/WiFiManager.h:16:25: fatal error: ESP8266WiFi.h: No such file or directory` error when I use tzapu/WiFiManager@^0.16.0 and the filenotfounderror when using https://github.com/tzapu/WiFiManager.git

sparkydave1981, 2021-09-01:
You are a legend!  Thanks for sharing this.  Now I just need to get all the settings sorted, but it is looking like I should get this working soon.  Thank you

closed 2021-09-01 by sparkydave1981