maragelis/ParadoxMqtt32 #24: Zone status not being sent to MQTT topics

sparkydave1981, 2021-09-01:
When viewing the MQTT topics in MQTTlens I am only getting data on `paradox32CTL/out` but no data on the zone topics such as `paradox32CTL/hassio/0/zone1`.  Also, even the 'out' topic does not seem to be showing all event.  Any tips?

sparkydave1981, 2021-09-01:
I have the below sensors in Home Assistant but only the one for Area0 is working.  I can arm / disarm the Paradox system via MQTT using this new interface, but only Area0 gives a status back, nothing from Area1.

```
sensor:
  - platform: mqtt
    name: paradox_alarm_status_area1
    state_topic: 'paradox32CTL/hassio/Arm/0'
    force_update: true
  - platform: mqtt
    name: paradox_alarm_status_area2
    state_topic: 'paradox32CTL/hassio/Arm/1'
    force_update: true
```

sparkydave1981, 2021-09-14:
Can someone please help me with this issue?  Perhaps I have done something wrong, but I don't know what it is...

qntris, 2021-11-14:
Hi @sparkydave1981 , the documentation is not quite right. Use the following for your sensor state_stopic:

paradoxdCTL/hassio/zoneX
where X is the zone number you have

sparkydave1981, 2021-11-30:
The above was for tracking the Area arm/disarm status rather than the zones.  My zone sensors are as per your example.  What do  you have for your Area sensors?