maragelis/ParadoxMqtt32 #25: Connection is slow and hangs very often

qntris, 2021-11-14:
Hi @maragelis,

Thank you for the great work you are doing to make our SP systems smarter than they are. I've decided to make the move from ESP8266 to ESP32 on my SP5500 board.
This is the ESP-32 30PIN that I have bought (3 pieces) from aliexpress:
[image]

I am uploading your code as it is, only changing the MQTT settings. The wifi connection is great, ping to the devices is perfect. The buck converter that I am using is passing 3.3v to the ESP32 board, as it should be. So far so good. However, I have the following issues:

- When I send a "stay" command for instance (with the proper payload), the Paradox board reacts (makes a sound) 3-4 seconds after that and starts arming_home. That is too much, compared to the ESP8266 board where each command is instant. Why does it happen?
- I don't see "Pending" status in HomeAssistant, compared to what I was seeing on the ESP8266. Is this expected?
- In about 20-30 minutes, the connection to the ESP32 device seems to hang (ping is still perfect) but it does not accept any commands, nor does it report zones movement.

I have erased all 3 boards, flashed them again and tried with each one of them - the result is absolutely the same. My setup is quite simple I think and really have run out of options.

Let me know if you want some further details from me, or if you want to do a join debugging session - my email address is:
ipuzunov at gmail dot com

Thank you in advance for your help and really hope that the ESP32 can prove to be the better option.

maragelis, 2021-11-15:
@qntris thanks for your kind words, I have seen this as well with my esp32. I am suspecting the PubSubClient on the esp32, its not recovering the mqtt connection ofter a timeout. The thing is I do not have the time to debug this, my day job is too demanding.

maragelis, 2021-11-16:
My hunch was correct so the pubsubclient is crashing together with the wifimanager, so I will upload a an update soon, like we say back to basics, I am running this on my installation and so far it looks stable.

qntris, 2021-11-16:
@maragelis great news, and I am glad you managed to figure it out. I suppose that's the reason for the "hanging" part, but what's the culprit for the slow response when issuing commands - several seconds for arming_home is definitely not expected. As I said, the WiFi connection is good on my end, ping is perfect, MQTT broker is stable... Do you suspect that the pubsubclient was to blame for that slowness as well?

Angel0ffDeath, 2021-11-23:
It is always good idea to add autoreboot in case MQTT cannot relive the connection. I had such problems with ESP266 and ESP32....

Although I continue to use ESP8266 for my alarm here are some tips:

void handleMqttKeepAlive() {
  if (!client.connected()) {
    long now = millis();
    if (now - lastReconnectAttempt > 5000) {
      lastReconnectAttempt = now;
      // Attempt to reconnect
      if (reconnect()) lastReconnectAttempt = 0;
    }
    if (millis() - lastConnected > reboot_timeout) ESP.reset();
  } else {
    // Client connected
    lastConnected = millis();
    client.loop();
  }
}

in reconnect before else delay(5000) add lastConnected=millis();
and somewhere in the beginning set reboot_timeout = 10*60*1000 (for instance)
This way if there is no connection more than 10 min ESP will self reboot.

Sorry the code is from ESP8266, but you got the idea and can quickly adapt it for ESP32 :)
Hope this will help to solve the issue....